ausocean/client #4: arduino/netsender: check for network in voltage alarm to prevent bricking

trekhopton, 2024-06-07:
The current ESP netsender software can get stuck in alarm mode if the configured voltage alarm value is reached.
The only way to get it out of alarm mode is by connecting it to a higher voltage, or re-flashing the ESP with another version of software.

Using a high input voltage is not always an option in the field, we experienced this at Rapid Bay. Re-flashing with another version is inconvenient. We also experienced this with devices we recycled and went from a 24V system to a PoE 24V system which has a slightly lower voltage (around 24V exactly).

I propose that we check for network connectivity and pull vars in the run cycle even when in voltage alarm. This way it can pull a new alarm recovery voltage value and we can recover the ESP from the alarm. This way we can use out default hotspot at least to recover the ESP (router won't usually work since it powers the ESP, but in the test rig case it still would have worked because it's connected to starlink)

scruzin, 2024-06-07:
I think the situation you describe is sufficiently rare that re-flashing the ESP is actually a reasonable resolution. In fact, in six years, the situation has happened only a couple of times. It is most often due to confusion between our standard 24V systems and (legacy) 12V systems. By standardizing on 24V systems (the plan!), this problem will never occur.

Furthermore, I don't think it is prudent to power up the router (and switch) during a low voltage alarm. If the voltage has been alarmed, we typically need to conserve every millijoule of energy. 99.9% of low voltage alarms are legitimate and powering up the network every time there is a voltage alarm will waste power and exacerbate an already serious situation.

Also, voltage is indeed low, the network equipment is not likely to be particularly reliable anyway.

If a smart phone is bricked, it is re-flashed (factory reset). I think we should do the same thing.

trekhopton, 2024-06-07:
It not JUST a reflash, we tried that. It's a reflash with a different version of netsender, then another reflash back the version you want. You mentioned that the ESP only clears vars when a NEW version is flashed.

I'm not suggesting we power on the network equipment but it should check if it can connect to the hotspot or configured network. Yes this is not a general purpose feature but it's useful for the AusOcean team when configuring and reconfiguing these.

The goal is not to brick the device in the first place, that's the last thing we want to do.

scruzin, 2024-06-07:
Rather than continually checking if the WiFi is on and fetching vars every cycle, a safer way to implement this might be to connect to WiFi and fetch vars _only after the ESP has rebooted due to alarm_. 

This means users would have 10 minutes (by default) to update the AlarmVoltage var.

scruzin, 2024-06-12:
Implemented in Pull Request #5

closed 2024-06-12 by scruzin