ausocean/client #30: Sensor Pins

ao-david, 2025-01-28:
Currently the DHT and SST are tied to pins 36 and 39. Since these pins are input only pins, they are incompatible with these sensors. It is proposed to instead move them to use pin 1 for the DHT, and pin 13 for the SST.

This does unfortunately use the UART TX pin which means that we would be sacrificing some of the useful IO for these sensors. It may also be possible to connect these devices onto a single pin since they both use the OneWire bus protocol. NOTE: This needs further investigation

ao-david, 2025-01-29:
I'm not going to make a pull request for the changes at this stage as it is only pin numbering, which may change in the near future with a PCB iteration, since the current implementation requires a custom board modification.

trekhopton, 2025-04-01:
I think it's too important to have access to the serial monitor while the DHT is plugged in to use pin 1 (TX). You have mentioned 12 is another alternative so we should go with that for DHT.