ausocean/client #33: temp-netsender: build failure with ESP8266

saxon-milton, 2025-02-01:
When building for `ESP8266` I get the following:
```
sketch/NetSender.cpp: In function 'void NetSender::wifiOff()':
NetSender.cpp:679:13: error: 'WIFI_MODE_NULL' was not declared in this scope
   WiFi.mode(WIFI_MODE_NULL);
             ^
exit status 1
'WIFI_MODE_NULL' was not declared in this scope
```
The relevant code is here 
```

void wifiOff() {
#ifdef ESP8266
  wifi_station_disconnect();
  bool stopped = false;
  for (int attempts = 0; !stopped && attempts < WIFI_ATTEMPTS; attempts++) {
    stopped = (wifi_station_get_connect_status() == DHCP_STOPPED);
    delay(WIFI_DELAY);
  }
  if (!stopped) {
    log(logError, "DHCP not stopping.");
    restart(bootWiFi, true);
  }
  wifi_set_opmode(NULL_MODE);
  wifi_set_sleep_type(MODEM_SLEEP_T);
  wifi_fpm_open();
  wifi_fpm_do_sleep(0xFFFFFFF);
#endif
  WiFi.mode(WIFI_MODE_NULL);
  delay(WIFI_DELAY);
#ifdef ESP32
#endif
  log(logDebug, "WiFi off");
}
```

`nonarduino.h` contains this macro, but `nonarduino.h` is only included for `__linux__`.

saxon-milton, 2025-02-01:
Should this be 
```
#ifdef __linux__
  WiFi.mode(WIFI_MODE_NULL);
  delay(WIFI_DELAY);
#endif
```
?

scruzin, 2025-02-02:
It should be:

```
#ifdef ESP32
  WiFi.mode(WIFI_MODE_NULL);
#endif  
  delay(WIFI_DELAY);
```

The delay should be called for all implementations.