ausocean/client #39: ESP: Not logging after first run

ao-david, 2025-03-27:
When inserted into the PCB, the ESP will stop logging after the first loop (after checking the alarm voltage).

When the ESP is floating (not in the PCB) it will continue to log the entire time.

This leads me to believe that there must be something connected to the same pin as the TX line that shouldn't be?

trekhopton, 2025-03-28:
It seems that this is because the DHT sensor was plugged into the GPIO1 (at least on the board marked "Faulty 2"). Otherwise I can't replicate this behaviour. Tried both v201 and v202 which both log fine to the serial monitor.

It should be noted that when the DHT was plugged into this board, garbage characters were being output. This is because of the clash on this board.

ao-david, 2025-05-20:
This was determined to be a hardware issue causing interference on the UART line. This has been fixed with the new standard revision of the controller PCB

closed 2025-05-20 by ao-david