ausocean/client #40: ESP: Brown outs

ao-david, 2025-03-27:
When bench testing, the ESP will look like it is reporting as the device is always green on the monitor page, however, the uptime would be only a few minutes (1-2), before seemingly restarting.

Looking at the requests on datablue, it looks like the ESP will read a zero value from the battery voltage pin, before restarting (indicated by a /vars request). However, the reboot reason reported (X13) would always be 0 (normal reboot).

[image]

NOTE: The battery pin is A4
Also, X10 which I think should report the analog value of the battery or the simulated battery seems to report -1 indicating an error.

trekhopton, 2025-03-27:
Just a note: -1 on X10 indicates an unconfigured AlarmVoltage, which is expected, as documented here:
https://docs.google.com/document/d/13rFl1nnFoeF-A8-bUJo0EKimct1zssA8jHgP2a1l2ec/edit?tab=t.0

> "A value of -1 means the device is not configured and the voltage check is disabled. If this is the case, first set AlarmVoltage to 1 (the smallest possible value that enables voltage checking) then update the device."

trekhopton, 2025-03-28:
I haven't been able to replicate this. I get network interruptions but this doesn't result in a vars request or restart. Tried both v201 and v202.
See my procedure here:
https://docs.google.com/spreadsheets/d/1s-RtVQsypZiDSbKoHLLKI5thidBOFtu4WtMAjxXg9Po/edit?gid=0#gid=0

and here:

https://docs.google.com/spreadsheets/d/1tznGQVQfwkljYBMUcS46yeEqW5dCduhPKZhFd0Ckmqw/edit?gid=0#gid=0

trekhopton, 2025-04-04:
I'm seeing this behaviour today. It's concerning and strange. Documenting as I go here:
https://docs.google.com/spreadsheets/d/1MkPD53byMZrbmB6ex5bHqmJ3ca13V4BK8Wafue-h05w/edit?gid=0#gid=0

It just did it while I was watching the logs:
```
DEFAULT 2025-04-04T05:35:01.385800Z /poll?vn=210&ma=88:13:BF:04:7F:D8&dk=1029384756&ut=10440&A4=883&A15=880&A26=871&A27=878&A14=0&A2=272
DEFAULT 2025-04-04T05:36:02.431092Z /poll?vn=210&ma=88:13:BF:04:7F:D8&dk=1029384756&ut=10501&A4=890&A15=883&A26=873&A27=881&A14=0&A2=295
DEFAULT 2025-04-04T05:37:02.401382Z /poll?vn=210&ma=88:13:BF:04:7F:D8&dk=1029384756&ut=10561&A4=879&A15=875&A26=867&A27=871&A14=0&A2=304
DEFAULT 2025-04-04T06:14:03.161449Z /poll?vn=210&ma=88:13:BF:04:7F:D8&dk=1029384756&ut=159&A4=880&A15=877&A26=0&A27=0&A14=0&A2=240
DEFAULT 2025-04-04T06:14:06.303035Z /vars?vn=210&ma=88:13:BF:04:7F:D8&dk=1029384756&ut=162
DEFAULT 2025-04-04T06:15:04.611162Z /poll?vn=210&ma=88:13:BF:04:7F:D8&dk=1029384756&ut=220&A4=879&A15=875&A26=0&A27=0&A14=0&A2=256
DEFAULT 2025-04-04T06:16:04.413595Z /poll?vn=210&ma=88:13:BF:04:7F:D8&dk=1029384756&ut=280&A4=882&A15=880&A26=0&A27=0&A14=0&A2=229

```

The most concerning thing is that I found it in a state this morning where the PCB alarm light was lit but I got no output from the serial monitor, no nav light was flashing and no attempts to connect to my hotspot. I reset the ESP and it start behaving normally until I noticed little brownouts.

trekhopton, 2025-04-04:
I feel like we need better debugging insights, is there a memory leak that is exceeding the heap? Maybe we can have more boot reasons. Maybe keeping a ESP log somehow would be good, perhaps sent via HTTP.

scruzin, 2025-04-07:
Let's rule out power supply and/or rig peripheral issues. I suggest testing the ESP32 in the lab on a PCB powered by a benchtop power supply.

trekhopton, 2025-04-07:
I have observed that partially covering the panels when the ESP is not reporting actually affects the system in such a way that allows the ESP to start reporting again. This indicates that the voltage is too high for something in the system.

Debugging log here:
https://docs.google.com/spreadsheets/d/1MkPD53byMZrbmB6ex5bHqmJ3ca13V4BK8Wafue-h05w/edit?usp=sharing

**12:27:21**	I opened the MEA after seeing that the rig hasn't reported since 11:17. The nav light is flashing. The ESP is not in alarm mode.

**12:30:28**	I looked at the router and saw that 4G was connected, refreshed the monitor page and the ESP had started reporting again. I think this is too coincidental. I feel like it might be the light sensor or something. I have unplugged the light sensor and I'll monitor it further. It seems so strange that it starts reporting as soon as I come to check on it. Perhaps me placing my laptop on the solar panels is decreasing the voltage which is allowing the ESP to behave properly..?

**12:40:06**	Okay it stopped reporting again. Last reported 12:37. (Coincidentally I'm inside now.) First thing I'm going to do is just put my laptop on the solar panels and see if it starts reporting again.

**12:46:22**	I place my laptop on the solar panel. Waiting.

**12:51:30**	The ESP started reporting again with nothing more than me placing my laptop on the solar panels. This leads me to believe the voltage is too high in the system and it's causing the ESP to behave strangely.

**12:53:02**	Took my laptop off the solar panels, let's see if it stops reporting again.

**13:19:54**	Hasn't reported since 12:55. This is pretty conclusive proof in my opinion that the voltage from the panels is causing the issue. The nav light / photometer PCB was unplugged FYI, so that wasn't causing the issue.

trekhopton, 2025-04-07:
I have further diagnosed it to the 5V line from the SCC cutting out every couple minutes. I replaced the SCC with another of the same model (AP20MT) and we'll monitor it tomorrow.

See the debugging log linked above, relevant observations below:

**16:32:41**	I probed the ESP's pin 5 which is the network power GPIO. It is 0 most of the time, then briefly changes to 0.6–1V. This seems to happen every minute — I set a timer and have observed this multiple times within that interval.
**16:41**	I had Shreyash help me video record the voltage of ESP pin 5 because I couldn't quite catch the reading. I thought I briefly saw 30V on the multimeter, but could be mistaken. It has a quick spike of (OVERLOAD), then 2.3V, 0.5V, 1.7V, then settles around 0.2mV again. I must have seen 3.0V — GPIO is between 0 and 3.3V. I'm going to monitor the ESP's 5V pin for more than a minute to see if it is restarting or if just the router pin is cycling. 5V pin on the ESP seems stable at 5.1V for over 2 minutes.
**16:55:34**	Oh, it started reporting again. That invalidates the test — I was hoping to record it while it was playing up. Nothing has changed about the rig this time; it’s just that time of day when it starts behaving again.
**16:59:22**	Measured 40V on the PV, 26.6V on the PCB input terminals.
**17:02:39**	It stopped again — likely because the sun came out more. Monitoring the 5V.
**17:06:11**	I monitored the 5V with the multimeter for a full 2 minutes and didn’t see any fluctuations.
**17:07:22**	PV is at 39V. It’s hard to know if the ESP only restarted once after the sun came out and is just not reporting because the router takes a while to initialise.
**17:09:54**	It started reporting again at 17:05.
**17:15:18**	It wasn't reporting. The sun came out and I finally witnessed the 5V line dip to 0.8V twice within a 2-minute period.
**17:16:43**	I'm going to unplug the ESP now and monitor the 5V from the USB-C to see if the behaviour persists.
**17:19:54**	I unplugged the ESP and monitored for 2 minutes. The voltage was stable but then dipped to 4.9V briefly, which is strange.
**17:23:23**	Watched it for another 2 minutes. It does this weird thing where it slowly falls over about 5 seconds to 4.2V, then rises back up to 5.1V. I'm going to read the SCC manual and see what the issue is.
**18:22:54**	I couldn't see anything in the manual. The only thing to check was the battery type, which was correctly set to white (lead acid). I swapped the SCC out on the MEA and we'll see how it goes tomorrow.

trekhopton, 2025-04-08:
The rig has now been reporting for 22.5 hours in sunlight without missing a minute. I think we can say that it was a faulty solar charge controller that had an unstable 5V line when PV was over 40V.
The cause of the fault in the charge controller is unknown. 
Closing this issue.

closed 2025-04-08 by trekhopton