ausocean/client #41: ESP: Alarm mode on startup

ao-david, 2025-03-27:
The ESP will also sometimes go into alarm mode briefly upon starting up (even when connected to a power supply).

This may be caused by the WiFi being on when trying to sample the pin, since the ADC 2 channels cannot be read whilst the wifi is on, and during startup, the WiFi is turned on, and the pins are read before it is turned off for the first time.

trekhopton, 2025-03-28:
I haven't been able to replicate this. Specifics on the PCB used would be helpful and what was plugged into it. How do you know it went into alarm mode. 
Please try to keep a detailed log of your debugging processes, like this:
https://docs.google.com/spreadsheets/d/1s-RtVQsypZiDSbKoHLLKI5thidBOFtu4WtMAjxXg9Po/edit?gid=0#gid=0

trekhopton, 2025-03-28:
I found your test process, linked here for future discussion: https://docs.google.com/document/d/1N37RM-TAhFbCnqkgjztFC24Ty4q1tzNSoys5C5uS3aE/edit?tab=t.0#heading=h.v5ujelh40ici

trekhopton, 2025-03-31:
I have found that the board can get into alarm mode (low voltage) at boot because of the first read being 0 for battery voltage. 
There's two issues it seems, one is that the 0 value is read so we are not turning wifi off it seems, and two is that it isn't coming out of the alarm despite a good voltage.

trekhopton, 2025-03-31:
The reason it wasn't coming out of alarm after a 0 read was because the recovery voltage was set too high, 25.5V when this test setup sits at 24.5V.

trekhopton, 2025-03-31:
The WiFi is turned on from the start at init.

```
// init should be called from setup once
void init(void) {
  // Disable WiFi persistence in flash memory.
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  Serial.begin(115200);
  pinMode(ALARM_PIN, OUTPUT);
  pinMode(NAV_PIN, OUTPUT);
  pinMode(STATUS_PIN, OUTPUT);
#ifdef ESP8266
  digitalWrite(STATUS_PIN, HIGH);
#endif
  delay(2000);

  // Get Config.
  readConfig(&Config);

  // Get boot info.
  XPin[xBoot] = Config.boot;
  Serial.print(F("Boot reason: ")), Serial.println(Config.boot);

  // Initialize GPIO pins, including power pins.
  initPins(true);

  // Save the formatted MAC address.
  byte mac[6];
  WiFi.macAddress(mac);
  fmtMacAddress(mac, MacAddress);
}
```

We have two options:
- Have WiFi on as the default and only turn it off when we need to read pins.
OR
- Have WiFi off as the default and only turn it on when we need to send a request. 

I think I prefer the latter since it may save power and seems more intuitive, turn on -> send -> turn off.

trekhopton, 2025-04-01:
Resolved in PR #43

closed 2025-04-01 by trekhopton