ausocean/client #54: arduino/netsender: need to sync client and server mode if the config request with LowVoltageAlarm mode is missed

trekhopton, 2025-05-20:
It seems that the ESP32 gets into a state where the mode = LowVoltageAlarm, but the alarm pin is not alarmed and the ESP is reporting ie. it's not actually in alarm mode.

I think this comes about if a config request is unsuccessful in communicating with datablue that the alarm has been cleared.

We should add a check in run: if !XPin[xAlarmed] AND Mode == LowVoltageAlarm THEN change the mode to Normal.

scruzin, 2025-05-20:
Alternatively, refactor the following:

```
    if (XPin[xAlarmed]) {
      if (XPin[xBat] < Config.vars[pvAlarmRecoveryVoltage]) {
        return pause(false, pulsed, &lag);
      }
      log(logInfo, "Low voltage alarm cleared");
      updateMode(mode::Normal);
      writeAlarm(false, true);
    }
```

as

```
    if (XPin[xAlarmed]) {
      if (XPin[xBat] < Config.vars[pvAlarmRecoveryVoltage]) {
        return pause(false, pulsed, &lag);
      }
      log(logInfo, "Low voltage alarm cleared");
      writeAlarm(false, true);
    }
    if (Mode != mode::Normal) {
      updateMode(mode::Normal);
    }
```

That way, we'll keep trying to reset the mode.

scruzin, 2025-05-20:
Also, updateMode should only update the global `Mode `if the config request is actually successful.

```
bool updateMode(const char* mode) {
  String reply;
  bool reconfig;

  if (wifiBegin() && request(RequestConfig, NULL, NULL, &reconfig, reply)) {
    log(logDebug, "mode=%s", mode);
    Mode = mode;
    return true;
  }
  log(logWarning, "Failed to notify service of mode");
  return false;
}
```

scruzin, 2025-05-24:
Issue #56 is the root cause of this issue. If we address Issue #56, this is no longer a problem.