ausocean/client #75: arduino/netsender: false 0V reading triggers lowVoltageAlarm

trekhopton, 2025-06-17:
While preparing an ESP32 v10017 (latest) for Port Noarlunga, we noticed it will read 0V for the A4 pin despite being powered by 24.5V and trigger a low voltage alarm. It seems like this happens after a network failure.

It seems to me that this particular ESP32 hadn't been fully configured. It had got the Wifi SSID etc., but it was still outputting serial monitor logs despite having no LogLevel set. This changed when I clicked 'Update' and it pulled the variables. I've added "Click Update" to the ESP32 configuration guide after the initial addition of the device: 
https://docs.google.com/document/d/1ckn8VODOjsjrjsqE5MlUllOYyLjGevJNzU9sfB6-TAA/edit?tab=t.0#heading=h.wsx2sz28e2p
Regardless I think it's a bug that a false 0V was triggering LowVoltageAlarm.

See the debugging log here:
https://docs.google.com/spreadsheets/d/1DQm1imfVCwMYzE9Yf9EsKlFRa9wkc4lBQbO4lDPwCaY/edit?gid=0#gid=0

scruzin, 2025-06-17:
I suspect that WiFi has been left on inadvertently, as we know that that the A4 pin yields unreliable values whenever WiFi is on.

ao-david, 2025-06-17:
There is no good reason. I suspect that when I initially decided on pins, I confused which ADC could be used with WiFi, as none of the ADC1 pins are used. When I designed revision B, I decided to leave the pins as they were to avoid causing confusion between versions more than necessary (and we seemed to have gotten around that problem).

scruzin, 2025-06-17:
PTAL at https://github.com/ausocean/client/pull/76

scruzin, 2025-06-23:
@trekhopton, please share the log file.

trekhopton, 2025-10-22:
I don't think we're seeing this issue in the latest versions. I'll close it now.