ausocean/client #84: arduino/temp-netsender: GPS support

scruzin, 2025-09-28:
It is proposed to add GPS support to the temp-netsender client (which has now outgrown its name since it measures considerably more than temperature).

Unlike the existing GPS support implemented by the Pi, it is proposed return latitude, longitude, and elevation as numbers, rather than as text.

ao-david, 2025-09-28:
This seems appropriate. note that Saxon has an old or for refactoring temp-netsender now that it has outgrown it's namesake

scruzin, 2025-09-28:
Regarding the returned values, we have a few choices:

1. Define some new X pins for latitude, longitude and elevation represented as integers.
2. Define some new X pins for latitude, longitude and elevation represented as real numbers.
3. Define a new pin type to represent single real numbers.
4. Define a new _collection_ (or _tuple_) type to represent a collection of two or more numbers (integers or real).

I lean towards option 4, although it requires server changes.

scruzin, 2025-10-01:
I recommend implementing the G (Group) pin as a `latitude,longitude,altitude` triplet initially. We can always extend the group and add extra items to group if we decide we need more info.

The semantics of a G pin is that all data values in the group share the same time, even if we represent the values as separate scalars.

scruzin, 2025-10-01:
That exactly how the ESP keeps track of time in offline mode. It's not super accurate, but it's better than nothing.

scruzin, 2025-10-06:
After discussing with @trekhopton, I opted to simply sending the whole NMEA sentence, which is associated with the T1 pin.

PTAL at https://github.com/ausocean/client/pull/85